BryggansBryggeri/electrical #30: Control 220V relays

jackonelli, 2019-07-01:
I added the hierarchical pins to complete the enclosure kicad but no actual functionality is in the pcb yet

jackonelli, 2019-08-16:
These need the exact same circuit as the SSR driver.
Transition to IC? E.g. http://www.ti.com/lit/ds/symlink/uln2003a.pdf

jackonelli, 2019-08-16:
- Den du länkade är Darlington arrayer, vilket inte är mosfets. Vilket kan vara fine om Raspberryn orkar driva dem

- Ålrajt. Men det finns säkert liknande med många mosfets?

- Det ska det göra
Den där det upp till 1.25mA per ingång, vilket är ganska mycket för Raspberryn
Man får inte dra mer än 16mA per pinne men totalen får inte överstiga 51mA. Det är lätt att glömma det om man börjar koppla på mer och mer saker

- Vi kanske ska gå direkt på en shift register och bara använda 3-pinnar?!?

- Gillar inte shift register. Har man problem med brus kan de vara opålitliga, men i2c io-expanders är nice https://se.farnell.com/microchip/mcp23017t-e-ss/i-o-expansion-16bitar-1-7mhz-ssop/dp/2851558 25mA per pin, 125mA totalt 16 styrbara pinnar i två 8-bitars register. Frågan är om 25mA räcker. Ville inte SSRerna ha typ 45mA, o skulle gissa på att relän vill ha något liknande

swebro, 2019-08-16:
[This](https://se.farnell.com/maxim-integrated-products/max4896etp/rel-drivkrets-8-kan-5-5v-tqfn/dp/2516973) seems like a nice little IC

jackonelli, 2019-08-16:
> [This](https://se.farnell.com/maxim-integrated-products/max4896etp/rel-drivkrets-8-kan-5-5v-tqfn/dp/2516973) seems like a nice little IC
> 
> EDIT: It even have Open-Load Detection, so we do not need to implement it ourself.

jackonelli, 2019-08-16:
Closed by mistake.

jackonelli, 2019-08-16:
> [This](https://se.farnell.com/maxim-integrated-products/max4896etp/rel-drivkrets-8-kan-5-5v-tqfn/dp/2516973) seems like a nice little IC
> 
> EDIT: It even have Open-Load Detection, so we do not need to implement it ourself.

But can it handle 24V needed for the SSR/relay. It says max Vcc 5.5V but it also says "designed for medium voltage applications up to 50V"

swebro, 2019-08-16:
Yes, it works very similarly to our current design if you look at page 11 of the datasheet.

swebro, 2019-08-16:
If I were to guess, the reason they chose 50V as max is that the internal source-drain capacitance is so large that the energy in the current rush when opening the transistor is enough to arc the junction

swebro, 2019-08-16:
The IC itself is just 5x5mm, so they do not have a lot of room for a wide transistor

jackonelli, 2019-08-16:
We are seriously considering i2c comm for an io expander:
https://www.farnell.com/datasheets/2243622.pdf

jackonelli, 2019-08-19:
Upgrade to opamp with more parallell inputs?

Here is a quad input variant (the tlc272 we currently use is only dual):
https://se.farnell.com/texas-instruments/lm324adr/op-amp-quad-low-power-smd-soic14/dp/8389233

jackonelli, 2019-08-19:
Inspiration:
https://www.abelectronics.co.uk/p/54/io-pi-plus

closed 2019-08-30 by jackonelli