BryggansBryggeri/electrical #32: Floating switch

jackonelli, 2019-07-05:
- [ ] Add to enclosure design
- [ ] Decide on connector
- [ ] Mount connector in enclosure

jackonelli, 2019-07-05:
Indicator is going to be tricky with serial wiring as we have planned.

jackonelli, 2019-08-19:
Simplest option is to have an implicit indicator:

Raspberry GPIO is firing without "AND error" but indicator for SSR is not HIGH. Then it should be the floating switch.

jackonelli, 2019-08-19:
If we want an actual indicator, which would be real nice. Then we could have the floating switch only wired to a Mosfet-style subcircuit which, when floating switch closed, allows SSR signal and indicator to close

JonathanFager, 2019-08-19:
Do we really need an indicator light for this? I agree it should not allow SSR to fire, but a light only for this seems like a feature creep

jackonelli, 2019-08-21:
Feature creep <3. Imagine having a red light go on as we are emptying the mash tun. Undaunted glory.

JonathanFager, 2019-08-30:
We could mount this in a stainless tube that is inserted into the kettles from the top. No need to drill holes.