BryggansBryggeri/electrical #34: Serial interface

jackonelli, 2019-08-14:
Will prevent major headaches to have this easily accessible even with active enclosure.

- [ ] Motherboard interface
- [ ] Interior wiring to sensor wall
- [ ] Panel mounted contact
- [ ] EPA-adapter to cable

Cable: https://se.farnell.com/adafruit-industries/954/usb-to-ttl-serial-cable-raspberry/dp/2215041

swebro, 2019-08-14:
I think that having a connector on the PCB so we can have a RJ-RJ cable between it and the connector on the case. [Here](https://github.com/stephane/libmodbus/wiki/Libmodbus-on-Raspberry-pi) is a library to use the internal UART to communicate over modbus, we just need a [tranciever](https://se.farnell.com/analog-devices/adm483earz/transceiver-rs485-250kbps-nsoic/dp/2377088?st=rs485) and a [rs485 line driver](https://www.google.com/search?q=rs485+bus+driver&source=lnms&tbm=isch&sa=X&ved=0ahUKEwjbidGeioPkAhUwwqYKHaSZC80Q_AUIESgB&biw=2327&bih=1180#imgrc=It-IkHaHQlSfnM:) if we want the RPi ti be the master

jackonelli, 2019-08-16:
Sadly, we need the uart pins for the VFD comms.

closed 2019-08-16 by jackonelli

jackonelli, 2019-08-29:
Terse, and for the rbpi-zero. But this might actually have something:
https://openpunk.com/post/5