BryggansBryggeri/electrical #40: Add Pressure/Volume sensor

JonathanFager, 2019-08-17:


jackonelli, 2019-08-17:
Not prioritised for first milestones. But nice to drill enclosure parts before mounting in brewery

JonathanFager, 2019-08-19:
There are 2 versions of this. One that delivers an analog signal of 0-5V or one that delivers a signal in the form of mA. Which is to be desired?
@swebro 

Also, what is the maximum pressure in kPA that can be expected? Needed to size the sensors correctly. @jackonelli

jackonelli, 2019-08-21:
With 2m fluid height: 121.429 kPa. So that is the absolute maximum.
I'm not sure about the exact height of the vessels

JonathanFager, 2019-08-21:
Boil 52+11.5cm (main+hoptrap)
Assume roughly same on mash

JonathanFager, 2019-08-21:
Boil main volume 158l.
Hoptrap 18l

JonathanFager, 2019-08-21:
70 cm depth gives roughly 7kPa?

jackonelli, 2019-08-21:
I assume we wont be operating the brewery in vacuum?

JonathanFager, 2019-08-21:
It will be very energy efficient to achieve boil of of DMS in a near vacuum.

B stands for Brownian motion

JonathanFager, 2019-08-21:
Now my calculations are not known to be correct. However: I assume that the sensors are calibrated to atmosoherical pressure. So they should be sized go only the added pressure of the water/wort hence 7kPa?

jackonelli, 2019-08-21:
Mais oui