BryggansBryggeri/electrical #42: Add Enclosure temperature sensor

JonathanFager, 2019-08-17:
A temperature sensor within the enclosure might be a good idea, if only as a temporary thing to monitor cabinet temeperatures and might even in thhe future sound buzzer or trigger E-stop if temperature exceeds say 70oC

jackonelli, 2019-08-17:
Great idea. I think part of it should be implemented. Triggering E-stop seems infeasible though.
How would we control the power to the contactor coil?

jackonelli, 2019-08-17:
We could of course block all output. But not the input?

JonathanFager, 2019-08-19:
I know for a fact that we have a DH11 Sensor lying around. Would be mounted directly on the PCB.
Ping @jackonelli to possibly implement this directly.

JonathanFager, 2019-12-09:
We now got a Ds18b20 in the enclosure.
This works well for all intent and purposes.
Will make a din-clip for ds18b20

JonathanFager, 2019-12-09:
First design iteration, hopefully it'll fit.
[image]

JonathanFager, 2019-12-09:
[ds18b20clip.zip](https://github.com/BryggansBryggeri/electrical/files/3940114/ds18b20clip.zip)

@jackonelli if you can, please start .gcode file extension on the printer.

JonathanFager, 2019-12-10:
Clip is printed. Probably in the box by the printer at ETA.

closed 2020-01-25 by jackonelli