BryggansBryggeri/electrical #43: Add Buzzer

JonathanFager, 2019-08-17:
Buzzer has two states continuous and pulsing.
Find buzzer and implement this so that both these states can be implemented. i.e Pulse for "Add hops" and Continuous for temperature is to high

jackonelli, 2019-08-17:
Buzzer is in box at ETA

jackonelli, 2019-09-06:
Ø: 26mm. Needs to be mounted on the outside of the enclosure.
Also: needs two separate signal lines: It has three pins, one neutral and two positive.
Positive 1: repeating beep
Positive 2: continuous beep.

Can this be driven low-side? I.e. is it symmetric?

jackonelli, 2019-09-06:
Or we can drive this with continuous signal only, and make beep patterns in software

JonathanFager, 2019-12-10:
Mounting-dinclip is printed. Probably in small box beside printer. 
Buzzer is on top of cabinet.

For now, not ideal to have inside of the cabinet, but will have to do for now.

What should the functionality be?