BryggansBryggeri/electrical #51: PE on SSR

jackonelli, 2019-09-06:
I think they are grounded in the DIN-rail, but there seems to be a dedicated screw for it, see page 6:

http://www.productselection.net/pdf/UK/RGC_2_3A.pdf

swebro, 2019-09-06:
You are correct, it is however normal to have a redundant connection in these kind of components. We could be content with how it is now, but it is better to also connect the ground screw

JonathanFager, 2019-10-10:
Today I crimped two wires for ground. However, they where a PITA to get to.
Measured the resistance from the SSR to different points on the enclosure and it was at max 0.7ohm. => More than good enough
@swebro Please move this back to To Do if you disagree, otherwise close the issue

swebro, 2019-10-17:
It should be good enough

closed 2019-10-17 by swebro