BryggansBryggeri/electrical #70: Analog Control VFD

JonathanFager, 2020-03-12:
- [ ] 0-10V/4-20mA?

https://literature.rockwellautomation.com/idc/groups/literature/documents/um/22b-um001_-en-e.pdf

jackonelli, 2020-03-19:
http://ww1.microchip.com/downloads/en/DeviceDoc/22187E.pdf
http://www.ti.com/lit/ds/symlink/lm2902-n.pdf
https://www.farnell.com/datasheets/187592.pdf

swebro, 2020-03-20:
We will have 4st 0-10V (implemted) and 4st 4-20mA (not implemented, reference design will be uploaded shortly) for general purpose control, in the beginning one of them will be used for the VFD until the MODBUS is implemented

swebro, 2020-03-20:
Perhaps we will need 24V for current loop control, so we should change the boost converter to a buck from the 24V

swebro, 2020-03-20:
[Reference design for current loop](https://www.falstad.com/circuit/circuitjs.html?cct=$+1+0.000005+10.20027730826997+50+5+43%0AR+96+80+16+80+0+0+40+4+0+0+0.5%0A207+96+80+96+-16+4+From%5CsDAC%0Aw+96+80+176+80+0%0Ar+176+176+176+256+0+180%0Aa+176+96+288+96+9+12+0+1000000+3.6587483686682742+4+100000%0Ag+176+256+176+288+0%0Aw+176+176+176+112+0%0Aw+176+176+368+176+0%0Ar+528+80+528+16+0+1000%0Aw+368+176+368+112+0%0AR+368+16+368+-64+0+0+40+24+0+0+0.5%0Af+288+96+368+96+32+1.5+0.2%0Aw+528+80+368+80+0%0Aw+368+16+528+16+0%0A207+528+16+528+-80+4+Loop%5Csout%0A)